Healthcare-Robotics/assistive-gym #1: TypeError: 'linkLowerLimits' is an invalid keyword argument for this function

GiZumRyue, 2019-12-18:
I have installed the env according to Install Guide, 
```
pip3 install git+https://github.com/Zackory/bullet3.git
git clone https://github.com/Healthcare-Robotics/assistive-gym.git
cd assistive-gym
pip3 install .
```
And run env_viewer.py. But there is a error like below.  
The version of assistive_gym is 0.100, and the pybullet is 2.6.0 .

```
Traceback (most recent call last):
  File "E:/MySrc/assistive-gym/env_viewer.py", line 19, in <module>
    observation = env.reset()
  File "D:\Anaconda3\envs\assistivegym\lib\site-packages\gym\wrappers\time_limit.py", line 25, in reset
    return self.env.reset(**kwargs)
  File "E:\MySrc\assistive-gym\assistive_gym\envs\scratch_itch.py", line 94, in reset
    self.human, self.wheelchair, self.robot, self.robot_lower_limits, self.robot_upper_limits, self.human_lower_limits, self.human_upper_limits, self.robot_right_arm_joint_indices, self.robot_left_arm_joint_indices, self.gender = self.world_creation.create_new_world(furniture_type='wheelchair', static_human_base=True, human_impairment='random', print_joints=False, gender='random')
  File "E:\MySrc\assistive-gym\assistive_gym\envs\world_creation.py", line 66, in create_new_world
    human, human_lower_limits, human_upper_limits = self.init_human(static_human_base, self.human_limit_scale, print_joints, gender=gender)
  File "E:\MySrc\assistive-gym\assistive_gym\envs\world_creation.py", line 89, in init_human
    human = self.human_creation.create_human(static=static_human_base, limit_scale=limit_scale, specular_color=[0.1, 0.1, 0.1], gender=gender, config=self.config)
  File "E:\MySrc\assistive-gym\assistive_gym\envs\human_creation.py", line 265, in create_human
    human = p.createMultiBody(baseMass=0 if static else m*0.1, baseCollisionShapeIndex=chest_c, baseVisualShapeIndex=chest_v, basePosition=chest_p, baseOrientation=[0, 0, 0, 1], linkMasses=linkMasses, linkCollisionShapeIndices=linkCollisionShapeIndices, linkVisualShapeIndices=linkVisualShapeIndices, linkPositions=linkPositions, linkOrientations=linkOrientations, linkInertialFramePositions=linkInertialFramePositions, linkInertialFrameOrientations=linkInertialFrameOrientations, linkParentIndices=linkParentIndices, linkJointTypes=linkJointTypes, linkJointAxis=linkJointAxis, linkLowerLimits=linkLowerLimits, linkUpperLimits=linkUpperLimits, useMaximalCoordinates=False, flags=p.URDF_USE_SELF_COLLISION, physicsClientId=self.id)
TypeError: 'linkLowerLimits' is an invalid keyword argument for this function
```

Zackory, 2019-12-18:
It looks like you have not installed the custom version of pybullet correctly.
You are using Anaconda, which may be using a different library directory than what pip3 is linked to.
Anaconda has liked installed the original pybullet library, which is missing joint limits.
I suggest trying a [python virtualenv](https://github.com/Healthcare-Robotics/assistive-gym/wiki/1.-Install) if you haven't already.

GiZumRyue, 2019-12-18:
> It looks like you have not installed the custom version of pybullet correctly.
> You are using Anaconda, which may be using a different library directory than what pip3 is linked to.
> Anaconda has likely installed the original pybullet library, which is missing joint limits.
> I suggest trying a [python virtualenv](https://github.com/Healthcare-Robotics/assistive-gym/wiki/1.-Install) if you haven't already.
> 
> For reference, you can find the linkLowerLimits argument in the pybullet python wrappers here: https://github.com/Zackory/bullet3/blob/master/examples/pybullet/pybullet.c#L8125

I've reinstalled customized pybullet, and it works!
Thanks for your help!

closed 2019-12-18 by Zackory

deepakraina99, 2020-12-27:
> I've reinstalled customized pybullet, and it works!
> Thanks for your help!

@GiZumRyue Did you reinstalled in conda env? How did you install it?

GiZumRyue, 2020-12-27:
> @GiZumRyue Did you reinstalled in conda env? How did you install it?

Hello! Maybe I just installed it in the conda following the readme and I found it in the main env of conda instead of the current env. I am sorry that i couldn't remember it very clearly, since it was quite a while from now.

deepakraina99, 2020-12-27:
> Hello! Maybe I just installed it in the conda following the readme and I found it in the main env of conda instead of the current env. I am sorry that i couldn't remember it very clearly, since it [...]

Okay, thanks. I will check.