Healthcare-Robotics/assistive-gym #2: screeninfo error

Neo-X, 2020-01-14:
File "/home/gberseth/playground/assistive-gym/assistive_gym/envs/env.py", line 9, in <module>
    from screeninfo import get_monitors
  File "/home/gberseth/playground/env/lib/python3.5/site-packages/screeninfo/__init__.py", line 1, in <module>
    from .common import Enumerator, Monitor
  File "/home/gberseth/playground/env/lib/python3.5/site-packages/screeninfo/common.py", line 10
    x: int
     ^
SyntaxError: invalid syntax

I found installing `pip3 install screeninfo==0.2` fix the issue.

Zackory, 2020-01-14:
Thank you for the update.
Installing Assistive Gym should install screeninfo 0.6.1, which I have found to work on Python 3.6.
Was screeninfo 0.6.1 throwing this issue?
Version 0.2 is 3.5 years old.

Neo-X, 2020-01-15:
Ahhh, the python3 virtual environment is using python3.5.

Zackory, 2020-01-15:
I have push an update to the setup.py file.
It should install screeninfo >= 0.6.1 if using Python >= 3.6, otherwise it will install screeninfo == 0.2.
Let me know if this works for you!

closed 2020-01-15 by Zackory