Healthcare-Robotics/assistive-gym #3: Model (keras) error

Elfits, 2020-02-18:
Brilliant work!!
I am running the `env_viewer.py` but I get one error one the model. I think it is due to the version of keras.
Which version of keras and tensorflow did you use?
FYI, here is the error log:
> E:\anaconda\python.exe F:/assistive-gym-test/examples/random_actions.py
pybullet build time: Feb 18 2020 16:57:18
Using TensorFlow backend.
2020-02-18 17:19:39.678134: I tensorflow/stream_executor/platform/default/dso_loader.cc:44] Successfully opened dynamic library cudart64_100.dll
WARNING:tensorflow:From E:\anaconda\lib\site-packages\tensorflow_core\python\compat\v2_compat.py:65: disable_resource_variables (from tensorflow.python.ops.variable_scope) is deprecated and will be removed in a future version.
Instructions for updating:
non-resource variables are not supported in the long term
Traceback (most recent call last):
  File "F:/assistive-gym-test/examples/random_actions.py", line 3, in <module>
    env = gym.make('FeedingPR2-v0')
  File "E:\anaconda\lib\site-packages\gym\envs\registration.py", line 156, in make
    return registry.make(id, **kwargs)
  File "E:\anaconda\lib\site-packages\gym\envs\registration.py", line 101, in make
    env = spec.make(**kwargs)
  File "E:\anaconda\lib\site-packages\gym\envs\registration.py", line 73, in make
    env = cls(**_kwargs)
  File "F:\assistive-gym-test\assistive_gym\envs\feeding_robots.py", line 5, in __init__
    super(FeedingPR2Env, self).__init__(robot_type='pr2', human_control=False)
  File "F:\assistive-gym-test\assistive_gym\envs\feeding.py", line 10, in __init__
    super(FeedingEnv, self).__init__(robot_type=robot_type, task='feeding', human_control=human_control, frame_skip=10, time_step=0.01, action_robot_len=7, action_human_len=(4 if human_control else 0), obs_robot_len=25, obs_human_len=(23 if human_control else 0))
  File "F:\assistive-gym-test\assistive_gym\envs\env.py", line 63, in __init__
    self.human_limits_model = load_model(os.path.join(self.world_creation.directory, 'realistic_arm_limits_model.h5'))
  File "E:\anaconda\lib\site-packages\keras\models.py", line 239, in load_model
    model = model_from_config(model_config, custom_objects=custom_objects)
  File "E:\anaconda\lib\site-packages\keras\models.py", line 313, in model_from_config
    return layer_module.deserialize(config, custom_objects=custom_objects)
  File "E:\anaconda\lib\site-packages\keras\layers\__init__.py", line 54, in deserialize
    printable_module_name='layer')
  File "E:\anaconda\lib\site-packages\keras\utils\generic_utils.py", line 139, in deserialize_keras_object
    list(custom_objects.items())))
  File "E:\anaconda\lib\site-packages\keras\models.py", line 1208, in from_config
    if 'class_name' not in config[0] or config[0]['class_name'] == 'Merge':
KeyError: 0

Zackory, 2020-02-18:
Hi Zhihao,

Here is a requirements.txt file with all of the Python libraries needed for a clean install of Assistive Gym.

[requirements.txt](https://github.com/Healthcare-Robotics/assistive-gym/files/4222110/requirements.txt)

Elfits, 2020-02-19:
Thanks!!
I updated keras to 2.3.1 and solved this problem.
However, after I run the `random-actions.py` in the `examples`, here are still errors:

> Traceback (most recent call last):
>   File "F:/assistive-gym-test/examples/random_actions.py", line 5, in <module>
>     observation = env.reset()
>   File "E:\anaconda\lib\site-packages\gym\wrappers\time_limit.py", line 25, in reset
>     return self.env.reset(**kwargs)
>   File "F:\assistive-gym-test\assistive_gym\envs\feeding.py", line 104, in reset
>     self.human, self.wheelchair, self.robot, self.robot_lower_limits, self.robot_upper_limits, self.human_lower_limits, self.human_upper_limits, self.robot_right_arm_joint_indices, self.robot_left_arm_joint_indices, self.gender = self.world_creation.create_new_world(furniture_type='wheelchair', static_human_base=True, human_impairment='random', print_joints=False, gender='random')
>   File "F:\assistive-gym-test\assistive_gym\envs\world_creation.py", line 66, in create_new_world
>     human, human_lower_limits, human_upper_limits = self.init_human(static_human_base, self.human_limit_scale, print_joints, gender=gender)
>   File "F:\assistive-gym-test\assistive_gym\envs\world_creation.py", line 89, in init_human
>     human = self.human_creation.create_human(static=static_human_base, limit_scale=limit_scale, specular_color=[0.1, 0.1, 0.1], gender=gender, config=self.config)
>   File "F:\assistive-gym-test\assistive_gym\envs\human_creation.py", line 265, in create_human
>     human = p.createMultiBody(baseMass=0 if static else m*0.1, baseCollisionShapeIndex=chest_c, baseVisualShapeIndex=chest_v, basePosition=chest_p, baseOrientation=[0, 0, 0, 1], linkMasses=linkMasses, linkCollisionShapeIndices=linkCollisionShapeIndices, linkVisualShapeIndices=linkVisualShapeIndices, linkPositions=linkPositions, linkOrientations=linkOrientations, linkInertialFramePositions=linkInertialFramePositions, linkInertialFrameOrientations=linkInertialFrameOrientations, linkParentIndices=linkParentIndices, linkJointTypes=linkJointTypes, linkJointAxis=linkJointAxis, linkLowerLimits=linkLowerLimits, linkUpperLimits=linkUpperLimits, useMaximalCoordinates=False, flags=p.URDF_USE_SELF_COLLISION, physicsClientId=self.id)
> TypeError: 'linkLowerLimits' is an invalid keyword argument for this function

Zackory, 2020-02-19:
It looks like you may have the original pybullet library installed.
You will want to install a customized pybullet here: https://github.com/Zackory/bullet3
Further details can be found hereL https://github.com/Healthcare-Robotics/assistive-gym/wiki/1.-Install

This custom pybullet library includes python wrappers for cloth simulation and human joint limits in pybullet.

Elfits, 2020-02-20:
Thanks!! It works now!
I think this issue (question) can be closed now and hope it would also be helpful for other beginners.
BTW, you and your team's work are so impressive! It is definitely a valuable project for deploying reinforcement learning with python bullet. I also visited your personal website and your experience and works are so brilliant! I will learn from you.  : )

closed 2020-02-21 by Zackory