Healthcare-Robotics/assistive-gym #7: Google Colab

kavikode, 2020-08-02:
Is it possible to run it on the Google Colab because Python Bullet does not seem to work there since it cannot connect to the X-Server?

Zackory, 2020-08-03:
Hi kavivode,

It is likely that you can train new assistive policies on Google Colab (see the [Wiki: Training New Policies](https://github.com/Healthcare-Robotics/assistive-gym/wiki/5.-Training-New-Policies)). Training a new policy does not require visualization or X-Server.

However, you will likely need Assistive Gym installed locally on a machine in order to visualize environments or trained policies.

For example, the command below won't work on Google Colab, since it requires brining up an interactive window through X-Server.
`python3 env_viewer.py --env "BedBathingSawyer-v0"`

Zackory, 2020-08-18:
However, it is possible to render a non-interactive simulation in Google Colab using an animated png.
Here is an example, which generates an animated png from a pybullet environment.

https://colab.research.google.com/drive/1u6j7JOqM05vUUjpVp5VNk0pd8q-vqGlx?authuser=1#scrollTo=7tbOVtFp1_5K

Zackory, 2020-12-29:
Assistive Gym now support Google Colab! Check out this [Wiki Page](https://github.com/Healthcare-Robotics/assistive-gym/wiki/7.-Google-Colab) for examples.

closed 2020-12-29 by Zackory