Healthcare-Robotics/assistive-gym #9: AttributeError: 'NoneType' object has no attribute 'BytesIO'

SamiraZare, 2020-08-09:
Hi, 

I keep getting this error while trying to train it after the first results. 

Updates 0, num timesteps 12800, FPS 1323 
 Last 64 training episodes: mean/median reward -152.7/-153.6, min/max reward -257.6/-87.9

Exception ignored in: <bound method SubprocVecEnv.__del__ of <baselines.common.vec_env.subproc_vec_env.SubprocVecEnv object at 0x7f15d7111e10>>
Traceback (most recent call last):
  File "/home/samira1/.local/lib/python3.6/site-packages/baselines/common/vec_env/subproc_vec_env.py", line 121, in __del__
  File "/home/samira1/.local/lib/python3.6/site-packages/baselines/common/vec_env/vec_env.py", line 98, in close
  File "/home/samira1/.local/lib/python3.6/site-packages/baselines/common/vec_env/subproc_vec_env.py", line 104, in close_extras
  File "/usr/local/lib/python3.6/multiprocessing/connection.py", line 206, in send
  File "/usr/local/lib/python3.6/multiprocessing/reduction.py", line 50, in dumps
AttributeError: 'NoneType' object has no attribute 'BytesIO'

Zackory, 2020-08-09:
Are you using Baselines 0.1.6?
You can install this version using:
`pip3 install git+https://github.com/openai/baselines.git`

SamiraZare, 2020-08-09:
Yes, I am using that. I used the tutorials here and installed all the requirements. It was working and training a couple of days ago. I installed it again, now it is giving me this error.

SamiraZare, 2020-08-09:
I fixed it! The issue was installing TensorFlow 1.14. It kept giving me this error when I installed Tensorflow 1.14. Now that I changed it to Tensorflow 2.2, it is working perfectly!

Thank you,
Samira

closed 2020-08-09 by SamiraZare