Healthcare-Robotics/assistive-gym #10: No friction on cloth

Yusufma03, 2020-09-21:
Hi, thanks for developing and sharing this interesting benchmark! I'm trying to further develop a cloth manipulation task based on your customized bullet environment. My problem is that the cloth is too slippery and seems to have no friction, so my PR2 robot will never be able to grasp it. I would really appreciate it if you could give me some suggestions. Thanks!

Zackory, 2020-09-22:
Hi Xiao. Yes, the process of grasping cloth is very tricky in simulation. This is primarily because only a few vertices of the cloth mesh end up in a robot's gripper, and it is challenging to get them to stay in between two parallel planes (for a parallel jaw gripper), without the mesh vertices getting pushed out of the gripper due to collision. We got around this in our dressing environment by creating an anchor (acts sort of like a stiff spring) between a few cloth vertices and the robot's end effector.

My suggestion would be to do something similar to this for grasping cloth. When the robot closes its end effector, loop through all the cloth mesh vertex positions and find the vertices that are inside or near the gripper fingertips. Then, create an anchor between these mesh vertices and the robot's end effector. When the robot open's its end effector up, you then delete all anchors between mesh vertices and the end effector. This require editing some of the pybullet code to create python wrappers for dynamically creating or deleting anchors, but it shouldn't be too much code. This would be similar to the custom c++ pybullet code used to create the cloth (which also creates anchors).

Yusufma03, 2020-09-23:
Hi Zackory. Thanks for the suggestions! I'll try that later. An additional question is that, when I load the .obj file with your customized loadCloth function, I might encounter some random segmentation fault errors. It might be resolved by just rerunning the script. 

Have you encountered this before?

DanielTakeshi, 2020-10-16:
Hi @Zackory  and @Yusufma03  just wanted to note that there's been more recent development on the official PyBullet code to support deformables. It is in the official documentation now.

Yusufma03, 2020-10-17:
Hi @DanielTakeshi , thanks for the notice. Actually I've tried the official PyBullet previously. The problem is that the loadSoftBody function is very slow and often stuck when I load my customized .obj file. It works with the loadCloth function implemented by @Zackory , although there are some random segmentation faults. 

I think this is not easily fixable and have switched to a different simulator. Thanks anyway! Issue closed.

closed 2020-10-17 by Yusufma03