Healthcare-Robotics/assistive-gym #12: Error with building custom bullet3

hzyjerry, 2020-12-02:
Hi,

I tried installing the custom bullet3 with pip3 on mac, and ran into this issue `error: command 'gcc' failed with exit status 1`.

So I tried cloning from the custom bullet3 repo, and building from there using `./build_cmake_pybullet_double.sh`. I ran into this following error:

```
/Users/jerry/Dropbox/Projects/AssistRobotics/bullet3/test/SharedMemory/./test.c:58:10: error: no matching function for call to 'b3PhysicsParamSetGravity'
                        ret = b3PhysicsParamSetGravity(command, gravx, gravy, gravz);
                              ^~~~~~~~~~~~~~~~~~~~~~~~
/Users/jerry/Dropbox/Projects/AssistRobotics/bullet3/test/SharedMemory/../../examples/SharedMemory/PhysicsClientC_API.h:326:20: note: candidate function not viable:
      requires 5 arguments, but 4 were provided
        B3_SHARED_API int b3PhysicsParamSetGravity(b3SharedMemoryCommandHandle commandHandle, double gravx, double gravy, double gravz, int body);
                          ^
```
Looks like `int body` is causing conflicts with the example code. Have you run into this issue before? Thanks for your insight.

hzyjerry, 2020-12-03:
Update: I was able to bypass the problems by installing on Ubuntu. The issue with Mac remains.

Zackory, 2020-12-03:
Hi Jerry!

What version of Python are you running on your Mac?
One suggestion is to switch to Python 3.6 (e.g. 3.6.5) using pyenv local install.

closed 2021-02-03 by hzyjerry