Healthcare-Robotics/assistive-gym #15: Issue with creating a new environment

deepakraina99, 2020-12-27:
I created a new environment by following the steps given in https://github.com/Healthcare-Robotics/assistive-gym/wiki/6.-Creating-a-New-Assistive-Environment.

I created an environment almost similar to BedBathingJacoEnv, but I get an error when I run it using the command python3 -m assistive_gym --env "TaskRobotEnv-v0":
```
gym.error.UnregisteredEnv: No registered env with id: TaskRobotEnv-v0
```

Zackory, 2020-12-28:
Hi Deepak. You may have installed the recent v1.0 of Assistive Gym. I have yet to update the environment creation example for v1.0 (as mentioned in the note at the top of the Wiki page you linked). First, can you check if you have installed v0.1 from this [branch](https://github.com/Healthcare-Robotics/assistive-gym/tree/v0.1)?

deepakraina99, 2020-12-29:
Hi @Zackory Thanks. v0.1 solved the issue.

closed 2020-12-29 by deepakraina99