Healthcare-Robotics/assistive-gym #16: Arm Manipulation Google Collaboratory

kavikode, 2021-01-27:
Can you please help me understand how to run the arm manipulation task shown in the right side, link: https://github.com/Healthcare-Robotics/assistive-gym/blob/9a308a8a989f179f1259824b579a063d610df110/images/human_preferences.gif?

How can we change the joint limitations of the human model for arm manipulation?

Thank you!

Zackory, 2021-01-27:
In general, you can test out the arm manipulation environment using the command
```bash
python3 -m assistive_gym --env "ArmManipulationSawyer-v1"
```

The figure you linked to is an example in which human preferences are disabled. This is done by commenting out this line: https://github.com/Healthcare-Robotics/assistive-gym/blob/main/assistive_gym/envs/arm_manipulation.py#L28
and then training a new robot control policy (see the Wiki: https://github.com/Healthcare-Robotics/assistive-gym/wiki/5.-Training-New-Policies).

The human arms have two types of joint limits. The first is pose-independent hard limits, which are set here: https://github.com/Healthcare-Robotics/assistive-gym/blob/main/assistive_gym/envs/human_creation.py#L217
The second type of joint limits that are applied to the arms are pose-dependent limits (put your hand behind your back and you will likely notice it is more difficult to fully bend your elbow). Pose-dependent joint limits are applied here: https://github.com/Healthcare-Robotics/assistive-gym/blob/main/assistive_gym/envs/agents/human.py#L129

Zackory, 2021-01-27:
For Google Colab, you can edit this notebook to use the environment `ArmManipulationSawyer-v1`
https://colab.research.google.com/drive/1PAY5HnLKRB-TBsPaevRr6myMfpVt_yzF?usp=sharing

Here is an example of training a new policy in Google Colab.
https://colab.research.google.com/drive/1P6zYEL2Pfq7GOCtmoLlRNGrKkPQTVAQF?usp=sharing

kavikode, 2021-02-03:
Thank you for your messages. How can we use this Assistive Gym Arm Manipulation example in ROS?

closed 2022-07-02 by Zackory