Healthcare-Robotics/assistive-gym #26: Hi, i can't run the env_viewer.py

987633259, 2022-04-30:
when I run the env_view.py in /assistive_gym, the error log is as follows,

```
/home/zing/anaconda3/envs/python3.6/bin/python /home/zing/anaconda3/envs/python3.6/assistive-gym/assistive_gym/env_viewer.py
/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:516: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:517: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:518: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:519: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:520: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:541: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:542: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:543: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:544: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:545: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:550: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
Traceback (most recent call last):
  File "/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/gym/envs/registration.py", line 158, in spec
    return self.env_specs[id]
KeyError: 'ScratchItchJaco-v1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zing/anaconda3/envs/python3.6/assistive-gym/assistive_gym/env_viewer.py", line 40, in <module>
    viewer(args.env)
  File "/home/zing/anaconda3/envs/python3.6/assistive-gym/assistive_gym/env_viewer.py", line 17, in viewer
    env = make_env(env_name, coop=True) if coop else gym.make(env_name)
  File "/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/gym/envs/registration.py", line 235, in make
    return registry.make(id, **kwargs)
  File "/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/gym/envs/registration.py", line 128, in make
    spec = self.spec(path)
  File "/home/zing/anaconda3/envs/python3.6/lib/python3.6/site-packages/gym/envs/registration.py", line 203, in spec
    raise error.UnregisteredEnv("No registered env with id: {}".format(id))
gym.error.UnregisteredEnv: No registered env with id: ScratchItchJaco-v1

Process finished with exit code 1
```

I don't know what the cause is. Can someone give some ideas? Thank you!

Zackory, 2022-05-01:
This may be an anaconda problem.
You might consider using pyenv.
"No registered env with id: ScratchItchJaco-v1" means that you have not installed Assistive Gym correctly, i.e. Python cannot find Assistive Gym.

987633259, 2022-05-03:
> This may be an anaconda problem. You might consider using pyenv. "No registered env with id: ScratchItchJaco-v1" means that you have not installed Assistive Gym correctly, i.e. Python cannot find Assistive Gym.

Thank you Zackory! But then i turn to a pyenv way, and I got a error report which may imply a version problem of the libraries. I can't check if this is caused by my OS.(my OS is Ubuntu 20.04)

/home/zing/.pyenv/shims/python3.6 /home/zing/assistive-gym/assistive_gym/env_viewer.py
/home/zing/env/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:516: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/zing/env/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:517: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/zing/env/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:518: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/zing/env/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:519: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/zing/env/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:520: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/zing/env/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
/home/zing/env/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:541: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/zing/env/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:542: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/zing/env/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:543: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/zing/env/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:544: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/zing/env/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:545: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/zing/env/lib/python3.6/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:550: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
pybullet build time: May  3 2022 15:07:44
Using TensorFlow backend.
Traceback (most recent call last):
  File "/home/zing/env/lib/python3.6/site-packages/ray/cloudpickle/__init__.py", line 39, in dumps_debug
    return dumps(obj, *args, **kwargs)
  File "/home/zing/env/lib/python3.6/site-packages/ray/cloudpickle/cloudpickle_fast.py", line 73, in dumps
    cp.dump(obj)
  File "/home/zing/env/lib/python3.6/site-packages/ray/cloudpickle/cloudpickle_fast.py", line 620, in dump
    return Pickler.dump(self, obj)
  File "/home/zing/env/lib/python3.6/site-packages/ray/pickle5_files/pickle5/pickle.py", line 485, in dump
    self.save(obj)
  File "/home/zing/env/lib/python3.6/site-packages/ray/pickle5_files/pickle5/pickle.py", line 601, in save
    self.save_reduce(obj=obj, *rv)
  File "/home/zing/env/lib/python3.6/site-packages/ray/pickle5_files/pickle5/pickle.py", line 689, in save_reduce
    save(func)
  File "/home/zing/env/lib/python3.6/site-packages/ray/pickle5_files/pickle5/pickle.py", line 601, in save
    self.save_reduce(obj=obj, *rv)
  File "/home/zing/env/lib/python3.6/site-packages/ray/pickle5_files/pickle5/pickle.py", line 689, in save_reduce
    save(func)
  File "/home/zing/env/lib/python3.6/site-packages/ray/pickle5_files/pickle5/pickle.py", line 576, in save
    rv = reduce(self.proto)
TypeError: can't pickle function objects

[... cut ...]
  File "/home/zing/env/lib/python3.6/site-packages/ray/tune/registry.py", line 166, in register
    self._to_flush[(category, key)] = pickle.dumps_debug(value)
  File "/home/zing/env/lib/python3.6/site-packages/ray/cloudpickle/__init__.py", line 47, in dumps_debug
    raise type(exc)(msg)
TypeError: ray.cloudpickle.dumps(<function <lambda> at 0x7f74e665f268>) failed.
To check which non-serializable variables are captured in scope, re-run the ray script with 'RAY_PICKLE_VERBOSE_DEBUG=1'.

Process finished with exit code 1

closed 2022-07-02 by Zackory