Healthcare-Robotics/assistive-gym #28: ('Observation ({}) outside given space ({})!' Error when trying to train model

zoe013evans, 2023-02-06:
Hi, 

I don't have any trouble getting the env viewer to work, but when I try to actually train the models I get issues. When I run the line: 

"python3 -m assistive_gym.learn --env "FeedingSawyerHuman-v1" --algo ppo --train --train-timesteps 100000 --save-dir ./trained_models/" 

I get an error to do with ray[rllib] that's something like:
      
 "    
      ray.exceptions.RayTaskError(ValueError): ray::RolloutWorker.par_iter_next() (pid=71524, ip=10.40.193.201)
        File "python/ray/_raylet.pyx", line 505, in ray._raylet.execute_task
        File "python/ray/_raylet.pyx", line 449, in ray._raylet.execute_task.function_executor
        File "/home/zoe/miniconda3/envs/assistivegymtwo/lib/python3.7/site-packages/ray/_private/function_manager.py", line 556, in actor_method_executor
          return method(__ray_actor, *args, **kwargs)
        File "/home/zoe/miniconda3/envs/assistivegymtwo/lib/python3.7/site-packages/ray/util/iter.py", line 1152, in par_iter_next
          return next(self.local_it)
        File "/home/zoe/miniconda3/envs/assistivegymtwo/lib/python3.7/site-packages/ray/rllib/evaluation/rollout_worker.py", line 332, in gen_rollouts
          yield self.sample()
        File "/home/zoe/miniconda3/envs/assistivegymtwo/lib/python3.7/site-packages/ray/rllib/evaluation/rollout_worker.py", line 706, in sample
          batches = [self.input_reader.next()]
        File "/home/zoe/miniconda3/envs/assistivegymtwo/lib/python3.7/site-packages/ray/rllib/evaluation/sampler.py", line 96, in next
          batches = [self.get_data()]
        File "/home/zoe/miniconda3/envs/assistivegymtwo/lib/python3.7/site-packages/ray/rllib/evaluation/sampler.py", line 223, in get_data
          item = next(self.rollout_provider)
        File "/home/zoe/miniconda3/envs/assistivegymtwo/lib/python3.7/site-packages/ray/rllib/evaluation/sampler.py", line 613, in _env_runner
          sample_collector=sample_collector,
        File "/home/zoe/miniconda3/envs/assistivegymtwo/lib/python3.7/site-packages/ray/rllib/evaluation/sampler.py", line 808, in _process_observations
          policy_id).transform(raw_obs)
        File "/home/zoe/miniconda3/envs/assistivegymtwo/lib/python3.7/site-packages/ray/rllib/models/preprocessors.py", line 187, in transform
          self.check_shape(observation)
        File "/home/zoe/miniconda3/envs/assistivegymtwo/lib/python3.7/site-packages/ray/rllib/models/preprocessors.py", line 68, in check_shape
          observation, self._obs_space)
      ValueError: ('Observation ({}) outside given space ({})!', array([ 0.74139816, -0.5481506 ,  0.13728762,  0.67066866, -0.0354565 ,
             -0.04076764,  0.73978668, -0.25926405, -0.50434195, -0.06268156,
             -0.94625477, -0.31422273, -0.55584883,  0.89838678, -0.86592606,
             -1.34115179, -1.86178359,  0.97019053,  0.03369198,  0.13699996,
             -0.21314327,  0.21102498,  0.05298619,  0.95248669,  0.        ]), Box(-1000000000.0, 1000000000.0, (25,), float32))
      Exception ignored in: <function ActorHandle.__del__ at 0x7f675cd07f80>

" 

I'm wondering if I'm using the wrong version of something like rllib or gym? 

I'm using python==3.7.10, gym==0.23.1, ray[rllib] == 1.3.0 

I've tried other version of these and still get the same error. I'm a bit unsure of how to fix. Any help or guidance would be greatly appreciated!! 

Thanks

zoe013evans, 2023-02-08:
Hi, 

Thanks for the reply! 

I've been unable to use Google Colab for a slightly different issue (which I also have when installing everything without Miniconda.) I run into the problem of not having tensorflow = 1.14.0, which I think is required by assistive gym (although I'm not sure), however I can't pip install this version onto my system unless I'm in a virtual environment because it's no longer available. I also can't seem to use this version of tensorflow in google colab. 

[image]

Any of the colabs that I use seem to have this issue. Installing using a Miniconda virtual environment has been the only way I've gotten the demos to run so far. 

Thanks again so much for the help

Zackory, 2023-02-08:
Okay, it looks like Google Colab has changed their Python version, so I'll need to update Assistive Gym to work for that.
However, on your local machine, I suspect the issue is miniconda.
Could you try pyenv instead? `pyenv install 3.7.10; pyenv local 3.7.10` then follow the installation: https://github.com/Healthcare-Robotics/assistive-gym#full-installation-to-editcreate-environments-using-a-python-virtual-environment

zoe013evans, 2023-02-09:
Thanks again for your help!
I tried it again with pyenv 3.7.10, but still ran into the same issue unfortunately

[image]