Healthcare-Robotics/assistive-gym #29: When I do "env = gym.make", I get an error "PicklingError: Could not pickle object as excessively deep recursion required.

kakeru58, 2023-02-16:
PicklingError: Could not pickle object as excessively deep recursion required.Hello, I am currently trying to use Assistive Gym with "ubunut20". 
When I do 
`env= gym.make`,  I get an error `PicklingError: Could not pickle object as excessively deep recursion required.`
If I use `sys.setrecursionlimit(3000)`, the error disappears, but `env = gym.make` does not finish after 10 hours.
`python3 -m assistive_gym --env "BedBathingSawyer-v1` also gives the same error. Is there any solution?

Thank you.

kakeru58, 2023-02-16:
After setting the python version to 3.6.9, pip install ray==1.11.0 worked fine.
I upload the environment.

# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
absl-py                   1.4.0                    pypi_0    pypi
aiosignal                 1.2.0                    pypi_0    pypi
argon2-cffi               20.1.0           py36h27cfd23_1  
assistive-gym             1.0                       dev_0    <develop>
astor                     0.8.1                    pypi_0    pypi
async-timeout             4.0.2                    pypi_0    pypi
async_generator           1.10             py36h28b3542_0  
attrs                     22.2.0                   pypi_0    pypi
backcall                  0.2.0              pyhd3eb1b0_0  
bleach                    4.1.0              pyhd3eb1b0_0  
ca-certificates           2023.01.10           h06a4308_0  
certifi                   2021.5.30        py36h06a4308_0  
cffi                      1.14.0           py36h2e261b9_0  
charset-normalizer        2.0.12                   pypi_0    pypi
click                     8.0.4                    pypi_0    pypi
cloudpickle               2.2.1                    pypi_0    pypi
commonmark                0.9.1                    pypi_0    pypi
cycler                    0.11.0                   pypi_0    pypi
dataclasses               0.8                      pypi_0    pypi
dbus                      1.13.18              hb2f20db_0  
decorator                 4.4.2                    pypi_0    pypi
defusedxml                0.7.1              pyhd3eb1b0_0  
distlib                   0.3.6                    pypi_0    pypi
dm-tree                   0.1.8                    pypi_0    pypi
entrypoints               0.3                      py36_0  
expat                     2.4.9                h6a678d5_0  
filelock                  3.4.1                    pypi_0    pypi
fontconfig                2.14.1               h52c9d5c_1  
freetype                  2.12.1               h4a9f257_0  
frozenlist                1.2.0                    pypi_0    pypi
gast                      0.5.3                    pypi_0    pypi
glib                      2.63.1               h5a9c865_0  
google-pasta              0.2.0                    pypi_0    pypi
grpcio                    1.43.0                   pypi_0    pypi
gst-plugins-base          1.14.0               hbbd80ab_1  
gstreamer                 1.14.0               hb453b48_1  
gym                       0.21.0                   pypi_0    pypi
h5py                      2.10.0                   pypi_0    pypi
icu                       58.2                 he6710b0_3  
idna                      3.4                      pypi_0    pypi
imageio                   2.15.0                   pypi_0    pypi
importlib-metadata        4.8.3                    pypi_0    pypi
importlib-resources       5.4.0                    pypi_0    pypi
ipykernel                 5.3.4            py36h5ca1d4c_0  
ipython                   7.16.1           py36h5ca1d4c_0  
ipython_genutils          0.2.0              pyhd3eb1b0_1  
ipywidgets                7.6.5              pyhd3eb1b0_1  
jedi                      0.17.0                   py36_0  
jinja2                    3.0.3              pyhd3eb1b0_0  
jpeg                      9e                   h7f8727e_0  
jsonschema                3.2.0                    pypi_0    pypi
jupyter                   1.0.0                    py36_7  
jupyter_client            7.1.2              pyhd3eb1b0_0  
jupyter_console           6.4.3              pyhd3eb1b0_0  
jupyter_core              4.8.1            py36h06a4308_0  
jupyterlab_pygments       0.1.2                      py_0  
jupyterlab_widgets        1.0.0              pyhd3eb1b0_1  
keras                     2.3.0                    pypi_0    pypi
keras-applications        1.0.8                    pypi_0    pypi
keras-preprocessing       1.1.2                    pypi_0    pypi
kiwisolver                1.3.1                    pypi_0    pypi
libedit                   3.1.20221030         h5eee18b_0  
libffi                    3.2.1             hf484d3e_1007  
libgcc-ng                 11.2.0               h1234567_1  
[... 72 lines cut ...]
tensorflow-estimator      1.14.0                   pypi_0    pypi
tensorflow-probability    0.7.0                    pypi_0    pypi
termcolor                 1.1.0                    pypi_0    pypi
terminado                 0.9.4            py36h06a4308_0  
testpath                  0.5.0              pyhd3eb1b0_0  
tifffile                  2020.9.3                 pypi_0    pypi
tk                        8.6.12               h1ccaba5_0  
torch                     1.10.2                   pypi_0    pypi
tornado                   6.1              py36h27cfd23_0  
traitlets                 4.3.3            py36h06a4308_0  
trimesh                   3.20.0                   pypi_0    pypi
typer                     0.7.0                    pypi_0    pypi
typing-extensions         4.1.1                    pypi_0    pypi
urllib3                   1.26.14                  pypi_0    pypi
virtualenv                20.17.1                  pypi_0    pypi
wcwidth                   0.2.5              pyhd3eb1b0_0  
webencodings              0.5.1                    py36_1  
werkzeug                  2.0.3                    pypi_0    pypi
wheel                     0.37.1             pyhd3eb1b0_0  
widgetsnbextension        3.5.1                    py36_0  
wrapt                     1.14.1                   pypi_0    pypi
xz                        5.2.10               h5eee18b_1  
zeromq                    4.3.4                h2531618_0  
zipp                      3.6.0                    pypi_0    pypi
zlib                      1.2.13               h5eee18b_0

Zackory, 2023-02-16:
Much appreciated! Thanks for posting the full environment list you have. Glad it is working for you now!

closed 2023-02-16 by Zackory