Healthcare-Robotics/assistive-gym #30: Issue running Assistive_Gym_Basics.ipynb colab notebook

rliu100, 2023-03-28:
Hi, I have been trying to run the Assistive_Gym_Basics.ipynb colab notebook. I am able to run the first cell, but get a `ModuleNotFoundError: No module named 'assistive_gym'` error when I try to run the second cell. 
[image]

Would this possibly be an issue with the python version? Thanks!

rliu100, 2023-03-29:
Update: I tried changing the python version to 3.6.15, and it still has the same error when running the second cell

Beatific0227, 2023-04-23:
I have the same problem. Also tried to run on local machine and it won't work

aaqib-ahmed-nazir, 2024-12-11:
Same issue